nathanmarlor/foxess_em #37: Not working in HA docker

djdeant, 2022-10-28:
## Version of the custom_component: v1.1.1

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

When using foxess_em on HA Docker on a Synology NAS, it does not appear to work.  See debug log that docker spits out.
Note that while is does say that No house load data available, this is incorrect, as the entity is valid and has the correct data in it.

## Debug log

2022-10-28 17:31:08.806 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up foxess_em platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 281, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/foxess_em/sensor.py", line 21, in async_setup_entry
await controllers["average"].async_refresh()
File "/config/custom_components/foxess_em/average/average_controller.py", line 66, in async_refresh
await self._model.refresh()
File "/config/custom_components/foxess_em/average/average_model.py", line 44, in refresh
await self._update_history(self._tracked_sensors[sensor])
File "/config/custom_components/foxess_em/average/average_model.py", line 51, in _update_history
await self._update_item(sensor.primary)
File "/config/custom_components/foxess_em/average/average_model.py", line 86, in _update_item
values_dict = [
TypeError: 'NoneType' object is not iterable
2022-10-28 17:32:29.288 WARNING (SyncWorker_2) [custom_components.foxess_em.battery.battery_controller] No house load data available

nathanmarlor, 2022-10-29:
Thanks @djdeant.

The integration needs at least two days of data to work, although looking at the exception it appears it isn't finding any... Can you send me a screenshot of your sensor.load_power (or whichever sensor you set as the house load) history please?

djdeant, 2022-10-29:
> Thanks @djdeant.
> 
> The integration needs at least two days of data to work, although looking at the exception it appears it isn't finding any... Can you send me a screenshot of your sensor.load_power (or whichever sensor you set as the house load) history please?

Hi @nathanmarlor,
Thanks for replying!

So, I used the foxess HA integration, which provides the load on "sensor.foxess_load_power" (screenshot of the graph attached)
[image].

I've had a look at the [foxess integration](https://github.com/macxq/foxess-ha/blob/main/custom_components/foxess/sensor.py), and I **believe** that this section is how the data is derived:

```yaml
class FoxESSLoadPower(CoordinatorEntity, SensorEntity):

    _attr_state_class = STATE_CLASS_MEASUREMENT
    _attr_device_class = DEVICE_CLASS_POWER
    _attr_native_unit_of_measurement = POWER_KILO_WATT

    def __init__(self, coordinator, name, deviceID):
        super().__init__(coordinator=coordinator)
        _LOGGER.debug("Initing Entity - Load Power")
        self._attr_name = name+" - Load Power"
        self._attr_unique_id = deviceID+"load-power"
        self.status = namedtuple(
            "status",
            [
                ATTR_DATE,
                ATTR_TIME,
            ],
        )

    @property
    def native_value(self) -> str | None:
        if self.coordinator.data["online"]:
            return self.coordinator.data["raw"]["loadsPower"]
        return None 

```

djdeant, 2022-10-29:
Ah wait a minute!  I can see in the logs that foxess_em is now logging into fox cloud and doing things, so maybe it is working now!

nathanmarlor, 2022-10-29:
Great, are the 14 energy management sensors populated and looking ok?

djdeant, 2022-10-29:
> Great, are the 14 energy management sensors populated and looking ok?

Yes, it appears so:
[image]

Can I ask, with the boost charge setting, does it also reset "time period 2" or does it store current settings and revert them?

djdeant, 2022-10-29:
It's OK, I think I found the HTTP POSTs in fox_cloud_service.py.  Looks like it will override anything I set - which should be fine in theory.

nathanmarlor, 2022-10-29:
Your house load still doesn't look quite right. The daily and peak loads should be different, plus the capacity at various times should be different - i.e. you'd expect the capacity at eco start to be higher than at end, as it's 4 hours later.

[image]

Can you send a screenshot of 48h of load power data please?

nathanmarlor, 2022-10-31:
@djdeant - this is potentially fixed in v1.2.1. I saw the same issues on my prod system but couldn't reproduce on development, despite the same ha.db?! Very odd. Give it a try, let me know :)