nathanmarlor/foxess_em #79: Force charging battery, even though Integration is turned off

booti101, 2022-11-15:
Apologies, this may be me doing something wrong as i have just installed this Integration.

I have found that even though the Integration is turned off, it is still turning on force charging of my battery each night. I have been to the FoxESS cloud page and made sure everything is disabled on that end, but it keeps turning it back on and setting the charge window to 24hrs (00:00 to 23:59) and the SoC (on grid) to 100%
This is resulting in my battery charging to full overnight, but then the house is pulling from the mains and also constantly charging the battery.

Solar forecast for the day is very low, but with the Integration turned off the day before, i am trying to work out why it has updated the Battery charge settings again and enabled a forced charge.
(I am on a flat rate tariff at the moment, but was trying to set the Management up ready for Eco 7 to supplied)

Running v1.3.4

Image showing the Energy Management all turned off in HA.
[image]

Image showing battery SoC. It only starts to discharge after 9am, once i change the settings again on FoxESS cloud manually
[image]

nathanmarlor, 2022-11-15:
Hi @booti101 - disable auto charge needs to be on (it's off in your screenshot) to prevent making any changes to the Fox settings (i.e. setting Min SoC, force charge periods...)

booti101, 2022-11-15:
@nathanmarlor Ah, my mistake. When you add the controls to a Dashboard from Integrations it put a toggle on the header (as you can see in the image). I thought this allowed you disable the entire Integration, but its just a slider.
Thanks for the clarification

closed 2022-11-17 by nathanmarlor