nathanmarlor/foxess_em #86: Accept sensor in place of Solcast information

Ashthos, 2022-11-18:
Hi,

Great integration, this has great potential to help me leave my FoxEss system alone!

Can you allow a sensor value to be used in place of the full-blown Solcast integration? I already have Solcast data being pulled into sensors (using the Solcast custom integration) and now foxess_em is also polling and using up lots of my daily credits.

Accepting a sensor as input would also allow me to run Solcast (or other) information through helpers and adjust the numbers in a manner that means they will be more accurate. 

Thanks.

nathanmarlor, 2022-11-18:
Thanks @Ashthos!

Unfortunately the integration requires the raw data coming from the Solcast API in order to build a model of battery vs. house load vs. forecast on a minute by minute basis. 

Two ways forward..

#1 - someone updates the Solcast integration to dump the raw values into a sensors attribute field, where this integration could pick them up.

#2 - I add whatever sensors you're missing from the Solcast integration into this (peak forecast etc.). Shouldn't be too difficult now it's polling, being parsed into a dataframe etc.

I'm happy with either tbh!

canton7, 2023-08-19:
It has the forecast as an attribute already?

[image]

nathanmarlor, 2023-08-19:
@canton7 I think OP meant to take the forecast as an input, so they could tweak/tailor the input before feeding into EM.

canton7, 2023-08-19:
I was responding to:

> #1 - someone updates the Solcast integration to dump the raw values into a sensors attribute field, where this integration could pick them up

I think solcast already does that, as shown in my screenshot?

nathanmarlor, 2023-08-19:
@canton7 I believe that was removed in a later version of the Solcast integration, although I'll happily be corrected.

canton7, 2023-08-19:
That's from the latest version, 3.1.6