nathanmarlor/foxess_em #111: Slow Charge - 1.5kW

C00K50N, 2022-12-05:
Just flagging this as an ongoing issue I'm facing, to take it out of someone else's thread I inadvertently hijacked...

When I'm using this integration my batteries no longer charge at a full 5kW, it's generally about 1.5kW.  Which means my batteries won't charge to full during the eco / cheap period, leaving me with less than 50% SoC by morning.

Since we last discussed this @nathanmarlor I have updated HA to 9.3 and Core to 2022.11.4, and am currently running your beta v1.5.1b6 but am still seeing the slow charge when I have the eco period set as 01:30 - 06:30.

Overview of logs:
[image]

I have generally tried to leave your Full Charge and Disable Auto Charge switches ON, and manually have my Fox Cloud settings to charge from grid 01:30 - 06:30 and this lets it charge properly.  Last night I hadn't realised my Fox Cloud settings were wrong (must have reset to 00:00 - 00:00 with switches off) but despite your Disable Auto Charge switch being ON the batteries did charge, albeit slowly.

[image]
[image]
This is the same behaviour I see when your Full Charge and Disable Auto Charge switches are OFF, which is how they return after a restart of HA and I have been known to forget to switch them back on.

There's definitely an argument that I should stop fiddling with everything, but it seems when I leave it alone the batteries won't charge at full power so I do need to get to the bottom of the issue.

Now that we're able to change the cheap times after install, I have changed them this morning to 09:00-09:10 just to be able to monitor what happens, and it's just charged as I typed at a full 5kW.  As it should have, but not helpful to identify what's going on when it's asked to do the same thing at 01:30!

When I changed FoxESS-EM time settings to 09:00-09:10
[image]
...with Min SoC set to 100% as expected.  

Reset to 00:00 at 09:10:
[image]

Let me know what other info you need from me to try and get to the bottom of this.  If it's something I'm doing wrong I'm happy to be pointed in the right direction, or if I can provide any more details for you to check anything at your end that would be great.

C00K50N, 2022-12-07:
Sounds good, thank you!  Yes, what you describe is I think what I've seen when I've still been awake after 01:30 and been able to check it: Min SoC has been changed to 99% but force charge switches haven't changed.  

I'd imagine a delay between calls might help; perhaps my hardware is slowing things down?  I'm running my HA on an old Pi 2.

If it's possible to achieve the same result by altering the schedule as required but leaving the Min SoC at 10% I wonder if we'd get a more reliable solution as that would better mimic what a user would do.  Then return the schedule to how it was set before being altered, rather than resetting it to 00:00-00:00 and switches off.

Are you also able to make your toggle switches (Full Charge, Disable Auto Charge, Boost Charge) "sticky" so they return to their previous values on a HA restart?  I'm seeing them switch off by themselves but I'm not sure what's causing it.

Finally, should Boost Charge be labelled as +1kWh rather than +1kW?  I presume it's to add a bit more charge?  If so it's energy rather than power, or am I mistaken?

Thanks for a brilliant integration!  Looking forward to this working hard come Spring when we no longer need full charge every night.

nathanmarlor, 2023-01-25:
Hi @C00K50N, lots of changes in -> https://github.com/nathanmarlor/foxess_em/releases/tag/v1.6.0b4

Will close for now, please let me know if these issues persist!

closed 2023-01-25 by nathanmarlor