nathanmarlor/foxess_em #152: Using grid power after overnight charge

andykorol1, 2023-01-22:
Hi, the last 2 mornings I've got up and the battery has charged as normal but it is staying on grid use. Both mornings I've gone to the foxess app and the attached screenshot were the settings. I've never put these in before. As soon as I correct them it all works again. Cheers 
[image]

nathanmarlor, 2023-01-22:
Thanks Andy. Can you attach some logs from around the time it should've stopped charging please? I'm guessing at a Fox Cloud connectivity issue.

andykorol1, 2023-01-22:
[image]

andykorol1, 2023-01-22:
`Logger: homeassistant
Source: helpers/event.py:336
First occurred: 6:30:00 AM (1 occurrences)
Last logged: 6:30:00 AM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/foxess_em/charge/charge_service.py", line 140, in _eco_end
    self._stop_listening()
  File "/config/custom_components/foxess_em/charge/charge_service.py", line 171, in _stop_listening
    listener()
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 308, in remove_listener
    _async_remove_indexed_listeners(
  File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 336, in _async_remove_indexed_listeners
    callbacks[storage_key].remove(job)
ValueError: list.remove(x): x not in list
`

andykorol1, 2023-01-22:
The last one is the at 6.30am when it should stop charging

nathanmarlor, 2023-01-22:
Not quite as I thought then. Can you confirm what version you're on please?

andykorol1, 2023-01-22:
Foxess energy management
1.5.2 

Inverter
Master 1.48
Slave 1.02
Manager 1.45

nathanmarlor, 2023-01-23:
Thanks. Does it continue to do this after a restart of HA? Has it happened all of a sudden? 

In short once the charging has finished we remove the listener to battery SoC (used to turn off charging when the desired SoC has been met) to prevent being notified about any other changes. It seems that HA/this component has lost the reference to that listener, which is very odd.

nathanmarlor, 2023-01-23:
Also, are there any errors around the off peak start time?

andykorol1, 2023-01-23:
Hi, It started randomly with no changes to HA but this morning appears to have sorted itself out. Didn't update anything so I'm guessing it got itself in a twist. Something I have noticed is the Capacity:next dawn time is 02:30 when my cheap window starts not actual daytime dawn. Is this correct? Also cheers for all the hard work that has gone into this.

nathanmarlor, 2023-01-23:
Thanks! It'll be much more useful in spring time when we start to see some proper solar!

Can you send me a screenshot of all the params please (like the attached)? Also a screenshot of the last 24h of house load. Dawn time should be the first time in the day that predicted solar is greater than predicted house load (or 12:00 midday if it doesn't exceed).

[image]

[image]

andykorol1, 2023-01-23:
[image]
[image]
[image]

nathanmarlor, 2023-01-23:
Odd. Have you set the timezone in HA? Container or HASS? 

What does the energy dashboard look like if you add Foxess as the solar prediction?

[image]

andykorol1, 2023-01-23:
[image]
[image]

It was setup without solar prediction. Have I done the correct settings in the other photo?

nathanmarlor, 2023-01-23:
Yep perfect, what does the energy graph look like with those inputs?

andykorol1, 2023-01-23:
[image]

nathanmarlor, 2023-01-23:
Now I'm super confused :)

I'm guessing the overnight load is an EV charger? Are you selecting to remove that through the Fox EM setup? I think it may be turning the load negative which then makes the diff > 0 (diff between load and solar) and confuses the calculation. I can add some extra guards around this (i.e. load is > 0) in the next release and see if it improves things for you.

andykorol1, 2023-01-23:
Yes it is an EV charger, which I've told Fox EM to ignore.

nathanmarlor, 2023-01-24:
@andykorol1 can you try the latest release please? Hoping this fixes your dawn calculations...

https://github.com/nathanmarlor/foxess_em/releases/tag/v1.6.0b2

andykorol1, 2023-01-24:
[image]

Looks like it's giving me a better time to dawn now

nathanmarlor, 2023-01-24:
Excellent! Will close this issue down.

BTW - the two capacity (dawn/eco end) entities can be deleted now, they were removed a few versions ago.

closed 2023-01-24 by nathanmarlor