nathanmarlor/foxess_em #205: Sensors showing unavailable

djdarlo, 2023-03-08:
After installing the integration it was initially working then after a restart all the sensors are showing unavailable.  Below are some logs from HA.

Home Assistant Core
No forecast data available
6:30:01 AM – (WARNING) FoxESS - Energy Management (custom integration) - message first occurred at March 7, 2023 at 6:57:36 PM and shows up 15 times
No house load data available
12:50:21 AM – (WARNING) FoxESS - Energy Management (custom integration) - message first occurred at March 7, 2023 at 6:51:48 PM and shows up 22 times
We found a custom integration tapo which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
March 7, 2023 at 7:24:15 PM – (WARNING) loader.py - message first occurred at March 7, 2023 at 7:24:15 PM and shows up 9 times
Forecast data could not be processed
March 7, 2023 at 7:13:47 PM – (WARNING) FoxESS - Energy Management (custom integration) - message first occurred at March 7, 2023 at 6:57:31 PM and shows up 3 times
Solcast API allowed polling limit exceeded
March 7, 2023 at 7:13:47 PM – (ERROR) FoxESS - Energy Management (custom integration) - message first occurred at March 7, 2023 at 6:57:31 PM and shows up 3 times
Solcast API allowed polling limit exceeded
March 7, 2023 at 7:13:47 PM – (ERROR) FoxESS - Energy Management (custom integration)
No house load data available
March 7, 2023 at 7:13:45 PM – (WARNING) FoxESS - Energy Management (custom integration) - message first occurred at March 7, 2023 at 6:57:24 PM and shows up 3 times
No forecast data available
March 7, 2023 at 7:00:40 PM – (WARNING) FoxESS - Energy Management (custom integration) - message first occurred at March 7, 2023 at 6:58:24 PM and shows up 4 times
No forecast data available
March 7, 2023 at 7:00:36 PM – (WARNING) FoxESS - Energy Management (custom integration) - message first occurred at March 7, 2023 at 6:58:24 PM and shows up 4 times

djdarlo, 2023-03-08:
Will try that now and feedback.

djdarlo, 2023-03-08:
Still getting this message following a restart....

Home Assistant Core
Setup of sensor platform foxess is taking over 10 seconds.
7:43:22 AM – (WARNING) Sensor
No house load data available
7:43:16 AM – (WARNING) FoxESS - Energy Management (custom integration) - message first occurred at 7:43:16 AM and shows up 2 times

nathanmarlor, 2023-03-08:
That's fairly normal - it's doing some chunky calculations in the background. Also you'll find either a warning for house load or forecast not being ready yet on startup, as you need both to be available to run the calcs.

Are you sure HA has fully started? The integration only runs once HA has fully started, as it relies on entity values which can be unavailable during startup.

djdarlo, 2023-03-08:
I have just double checked and it now looks fine :) Sorry for messing you around :(  Any idea why it wasn't working earlier?

[image]

nathanmarlor, 2023-03-08:
Good news!

It's because of this -> https://github.com/nathanmarlor/foxess_em/blob/main/custom_components/foxess_em/forecast/forecast_controller.py#L128-L134

If it can't refresh the forecast (probably due to hitting the Solcast API limit) it then won't setup the next refreshes, so stays in a broken state until restart. I'll fix it for the next release :)

nathanmarlor, 2023-03-16:
Fixed in #212 and #213

closed 2023-03-16 by nathanmarlor