nathanmarlor/foxess_em #235: Calendar functionality not working since BST change

andrewmooreio, 2023-03-26:
## Version of the custom_component

1.7.2

## Describe the bug

Since the clocks went forward last night, the calendar functionality of the integration is no longer working. There are no calendar entries for the daily charge and the following error shows up on the calendar page:
[image]

First thing I did was update to 1.7.2 because there was a fix relating to timezones, but that doesn't seem to have resolved the issue.

I'm seeing a recurring error in the logs, which I've pasted below.

## Debug log

```text

Logger: aiohttp.server
Source: custom_components/foxess_em/calendar.py:62
Integration: FoxESS - Energy Management ([documentation](https://github.com/nathanmarlor/foxess_em), [issues](https://github.com/nathanmarlor/foxess_em/issues))
First occurred: 11:35:51 (8 occurrences)
Last logged: 11:44:11

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 67, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 227, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 234, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 146, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/calendar/__init__.py", line 446, in get
    calendar_event_list = await entity.async_get_events(
  File "/config/custom_components/foxess_em/calendar.py", line 62, in async_get_events
    summary += f"Import: {str(round(values['import'], 2))} / "
KeyError: 'import'

```

nathanmarlor, 2023-03-26:
Apologies - I half expected something to break even though I'd been super careful with datetimes. All sorted in #236

closed 2023-03-26 by nathanmarlor

andrewmooreio, 2023-03-26:
I've updated to 1.7.4 and the issue still persists:

```Logger: aiohttp.server
Source: custom_components/foxess_em/calendar.py:56
Integration: FoxESS - Energy Management ([documentation](https://github.com/nathanmarlor/foxess_em), [issues](https://github.com/nathanmarlor/foxess_em/issues))
First occurred: 14:59:55 (6 occurrences)
Last logged: 15:01:24

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 433, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 67, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 227, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 234, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 146, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/calendar/__init__.py", line 446, in get
    calendar_event_list = await entity.async_get_events(
  File "/config/custom_components/foxess_em/calendar.py", line 56, in async_get_events
    charge = str(round(values["total"], 2))
KeyError: 'total'```

nathanmarlor, 2023-03-26:
Hmmpphhh.

Can you try removing the integration and re -adding please? At a guess the cached schedule still has some timezone weirdness going on.

andrewmooreio, 2023-03-26:
No problem. I'll probably end up doing it tomorrow, purely because I'm at 8 API hits on Solcast today. Re-installing will take me to the limit.

I'll let you know! Thanks.

nathanmarlor, 2023-03-26:
No problem. Back to fully functioning on my HA after reloading so I'm sure it'll sort yours too. Hopefully that's the last of the timezone gremlins 😂🤦

andrewmooreio, 2023-03-27:
All good after a clean reinstallation 👍