nathanmarlor/foxess_em #249: Daytime Charging?

oilybits, 2023-03-29:
Hi Nathan,
Many thanks for this great programme.  I installed HA and FoxESS EM in the middle of Feb this year and I have stayed up late each night to see how the Energy management system works. After 6 weeks of patient viewing, we finally hit pay dirt this morning when it was fairly clear that the forecast sunshine would not be sufficient to keep my batteries topped off.
According to my statistical guesstimates, my solar system should become pretty much self-sustaining from 20 March onwards so I was expecting to shut the system down until next November…..
So, at the allotted time this morning/last night, the batteries started charging all by themselves for the very first time and I was mightily impressed.
I’m a retired mechanical engineer who is loath to make too many tweaks on parameters without fully understanding the implications however I noticed that throughout the day, the battery has been charging in small doses to keep the battery topped up on daytime electricity (I’m in the UK).
I’ve attached two screen shots to show the grid consumption during the day and another to show the switch settings on the Energy Management system.  These settings are unchanged since February so I can only assume that the first use of the auto top up facility has somehow triggered this.  Note that I have the custom charge profile ‘set’ as this is the only way of keeping the system ‘on’ without doing a full charge, a top-up or disabling auto charge.
Are there some extra tweaks I need to make to prevent this from happening?
Thank you in anticipation,
Alan
[image]

nathanmarlor, 2023-03-30:
Thanks @oilybits 

A little concerned your next dawn time is showing as yesterday.

Can you include screenshots of 48h of load power, grid consumption and SoC please?

oilybits, 2023-03-30:
Hi Nathan,

Thank you for the speedy come-back.

I've attached 48 hr trends as requested.  I managed to turn off the 
trickle charge by overriding the HA settings in Foxesscloud.

As I write this, 'capacity: next dawn time' is showing 9 hours which 
seems about right.  For reference, the battery stack is 12.3 kWh.

Best regards,

Alan

nathanmarlor, 2023-03-31:
@oilybits where have you attached them? :)

oilybits, 2023-03-31:
[image]

oilybits, 2023-03-31:
Hi Nathan,
I responded directly to the mail from my inbox with the files appended to that.  I can see that they don’t appear on GitHub so I just uploaded them……
Cheers,
Alan

oilybits, 2023-04-06:
Hi Nathan,
Any thoughts on this?
Cheers,
Alan

nathanmarlor, 2023-04-06:
Hi Alan, ah I think I can see what's happening here. At 00:30 the integration started a force charge and set the min SoC (on grid) to ~50% (to prevent further discharge if load shedding etc.), however it appears to have failed to have set it back at the end of the off-peak period. This has resulted in grid usage until PV kicked in, and grid usage after sunset and any additional charge had been used. 

Are you using the Cloud or Modbus integration?

oilybits, 2023-04-07:
Hi Nathan,
I believe I'm using the Cloud integration.  (I gave HA my Foxcloud login details).
Cheers,
Alan

oilybits, 2023-04-10:
Hi Nathan,
Have you eaten all your Easter eggs, yet?
I upgraded to v 1.7.4 yesterday.  Overnight, the batteries fully recharged and maintained 100% until end of cheap rate.  Is this related to the upgrade or my previous issue?
Thanks,
Alan

nathanmarlor, 2023-09-05:
@oilybits are you still experiencing issues or can this be closed?

oilybits, 2023-09-05:
Hi Nathan,
I suspect the issue is still open.  My inverter (4 months old) chucked its hand in at the end of June.  The replacement is a later model which has a different wifi interface and I'm unable to make my rasp Pi talk to it......
I hope to re-install HA and have another bash at it shortly.  I'll let you know once I'm fully up and running again.
Thanks,
Alan

oilybits, 2023-10-02:
Hi Nathan,

Following premature failure of my inverter, I finally got my new inverter to talk to HomeAssistant (the process was somewhat beleageured by my low frequency wifi connection).

The 'updated' replacement inverter has no LAN connection and the usb port is taken up by an Anatel wifi stick.  For the moment, I see no possibility to communicate directly with the inverter through the cloud integration so I won't be able to explore this issue further.

Please close the query and I'll see if I can glean some more information from Will Eccle's Foxess webpage.

Thank you,

Alan

closed 2023-10-02 by oilybits