nathanmarlor/foxess_em #256: Expand Solcast API feature to allow two Solcast sites (e.g. split or dual solar arrays)

StealthChesnut, 2023-04-05:
**Is your feature request related to a problem? Please describe.**
Support two Solcast sites (e.g. for a split or multiple solar arrays)

**Describe the solution you'd like**
When setting up the Solcast part of the flow, allow input of two different solcast sites. A free/home user account allows upto 2 sites, so there are 2 different site IDs.
If there are 2 Solcast sites then the data for both will need to be added together - there doesn't appear to be a single API call to do this their end.

**Describe alternatives you've considered**
Use forecast.solar data from within Home Assistant?

StealthChesnut, 2023-04-05:
Closing this - raised prematurely! Functionality is already present!

closed 2023-04-05 by StealthChesnut

screenagerbe, 2023-04-08:
How exactly?
Can't see it tbh