nathanmarlor/foxess_em #283: Possibility of using inverter lan port?

littlefeltfangs, 2023-05-10:
Is there any possibility of extending this integration to work with the inverter's own lan port provided modbus? Or is this a restriction on the inverter end?

I am currently using the FoxESS - Modbus integration and the built in lan to read data from the inverter, which is working well. I've not had any luck getting either integration to work through fox cloud and I'm not confident messing with the wiring in my inverter, so being able to use the built in LAN would be really valuable if doable.

nathanmarlor, 2023-05-11:
Unfortunately not as the register addresses needed to control charge periods haven't yet been mapped for the LAN port.

canton7, 2023-05-13:
... And I strongly doubt they ever will be. We know for sure that the set of registers exposed over the lan port is more restricted than those exposed over RS484. People have searched the entire address space looking for them, and come up blank.