nathanmarlor/foxess_em #325: Almost no sensors have any data

tekhiegit, 2023-08-01:
Hi, this is a great integration to control the charging of my fox system, but beyond having the buttons on my dashboard I've not been able at any point in the past to get it to populate the calculated figures. 

I believe that is all setup correctly, I'm connected via modbus and am able to control charging and have it do it at set times, or does get forecast data, but that's it. Non of the other sections of data get filled out. I've tried multiple restarts of home assistant and this integration. But I guess I'm doing something wrong somewhere. Can anyone help?

[image]

Snommas, 2023-08-05:
@tekhiegit Hello, I'm having the same problem. 
@nathanmarlor  Nathan, I am running FoxESS_Modbus already and it works well, thanks. I'm connecting to the inverter with RS485 to USB serial adapter. After installing FoxESS_em the polling indicaters on the adapter never restart, probably why most of the enties are unavailable?
There is a discussion on here that says the two integrations can co-exist, I've had multiple attempts with no luck. May I ask for your thoughts on this.
Thanking you

nathanmarlor, 2023-08-05:
Hi @tekhiegit - it looks like it's unable to calculate the average house load. Can you include logs + a screenshot of the last 48h of house load please.

@Snommas the two can co-exist, albeit the modbus integration currently supports far more inverters (UDP etc.). EM will catch up eventually but currently it's W610, Waveshare style adapters only.

Snommas, 2023-08-05:
Thank you for your reply and hard work on this.
What's chance of me importing some of the fuctionality of _EM into Modbus..? Just a thought :-)
Thanks again.

Thebig24fan, 2023-09-07:
Great job on the modbus but having same unknown issues with this bit.  Great idea on this auto charging idea but cant get it to work thanks for the help sort this bug out

nathanmarlor, 2023-09-07:
@Thebig24fan logs and screenshots of 48h load data please.

Thebig24fan, 2023-09-08:
Ok 👌 which log to you want and how do i add them here thanks some of the fields have now populated with data  most with 0 and but charging needed  and eco start etc still unknown and  rest dont know how accurate it is ive tried reboot ha host and reinstall the em integrate etc still the same heres some screenshots 
[image]
[image]
[image]

siobhanellis, 2024-05-10:
I am also having the same problem.

I do see an error in the log file that says 

Error (Syncworker_50) [custom_components.foxess_em.battery.battery_controller] ValueError("Invalid literal for int() with base 10: '60.0'")