nathanmarlor/foxess_em #326: Cannot configure Integration in Home Assistant - Connection Timeout for ModBus

r-amado, 2023-08-03:
## Version of the custom_component
1.8.0

## Configuration
no configuration yet, still trying to add the integration.

## Describe the bug
Inverter already being used through ModBus TCP with your Integration [nathanmarlor/foxess_modbus]
When trying to configure the Integration gives error connecting to the inverter, on step 2/4.
[image]

Is it possible to have connection timeout due to having already an integration connected to the inverter?
If so, does this integration gives all the information in realtime like the other one you have published?
[nathanmarlor/foxess_modbus]

## Debug log
2023-08-03 12:19:41.737 ERROR (SyncWorker_6) [pymodbus.logging] Connection to (192.168.68.152, 502) failed: timed out
2023-08-03 12:26:19.202 WARNING (MainThread) [custom_components.foxess_em.config_flow] ModbusIOException()

nathanmarlor, 2023-09-05:
@r-amado this part of EM functionality will be overhauled shortly to interface directly with the modbus integration.

markhaines, 2024-07-09:
It looks like I have same issue - using a Waveshare RS232/485/422 to RJ45 Ethernet Module, I have this integrated with HA using the FoxESS Modbus connection (over TCP) fine.  Won't let me connect using this integration, I have 'multi-host' option enabled (although not sure whether the related parameters are correct).  Is it still the case that this direct connection won't work and I'll need to stick to the cloud connection for this integration for the time being?

bmgmac, 2024-09-23:
Hi
I too am having this option and unable to connect via an EW11 RTC to TCP device and it gives me the same error but different debug log. 

[image]

This is my debug log for an attempted connection. 

2024-09-24 00:39:10.372 DEBUG (MainThread) [custom_components.foxess_em.fox.fox_modbus] Connecting to modbus - ({'connection_type': 'Modbus TCP', 'host': '192.168.0.88', 'port': '502'})
2024-09-24 00:39:10.372 DEBUG (MainThread) [custom_components.foxess_em.fox.fox_modbus] Reading register: (40002, 1, 247)
2024-09-24 00:39:10.510 DEBUG (MainThread) [custom_components.foxess_em.fox.fox_modbus] Closing connection to modbus
2024-09-24 00:39:10.511 WARNING (MainThread) [custom_components.foxess_em.config_flow] ModbusIOException('[Input/Output] Error reading registers: Exception Response(131, 3, IllegalAddress)')

Any help would be gratefully received.
Many thanks