nathanmarlor/foxess_em #340: Strategy Mode

Joshuaharry120, 2023-10-20:
Please can you intergrade the ability within foxess_em to allow discharging to grid. - seems this can only be done when there is a strategy, let it run same as forge charge, it'll force charge until you tell it to stop

This has recently been added to the fox via the latest manager version 1.70.

[image]

nathanmarlor, 2023-10-20:
@Joshuaharry120 see https://github.com/nathanmarlor/foxess_modbus/discussions/440#discussioncomment-7314557

Joshuaharry120, 2023-10-20:
> [](https://github.com/nathanmarlor/foxess_modbus/discussions/440#discussioncomment-7314557)

was hoping for a non-Modbus way ? - not likely ?

Joshuaharry120, 2023-10-24:
It Seems You can set it using the API.

https://www.foxesscloud.com/i18n/en/OpenApiDocument.html#psuk