nathanmarlor/foxess_em #347: Api calls failing with 406

FingalP, 2023-12-20:
## Version of the custom_component

1.8.0

## Configuration

```yaml
sensor:
  - platform: foxess
    username: Fingal
    password: {my_password}
    deviceID: {my_device_id}```

## Describe the bug

Hi, I'm trying to use the `foxess_em.stop_force_charge` service via the cloud API, and I'm getting a 406 Not Acceptable error.
![image](https://github.com/nathanmarlor/foxess_em/assets/31482158/b79a0905-0c62-4978-9402-6c6c277f49f8)

I'm also unable to go through the setup process again, as it says `Error - please check username and password` even though I have double and triple checked my password (and reset it to the same thing again).
## Debug log

<!-- To enable debug logs check this https://www.home-assistant.io/components/logger/ -->

```text

2023-12-20 10:54:52.812 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration foxess_em which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-12-20 10:54:52.813 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration foxess which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-12-20 10:54:55.051 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'homeassistant_alerts', 'energy', 'mobile_app', 'integration', 'conversation', 'onboarding', 'analytics', 'input_number', 'input_boolean', 'stt', 'timer', 'script', 'search', 'shopping_list', 'input_datetime', 'counter', 'diagnostics', 'tag', 'default_config', 'automation', 'input_select', 'input_button', 'lovelace', 'media_source', 'tts', 'hardware', 'assist_pipeline', 'wake_word', 'hacs', 'forecast_solar', 'blueprint', 'foxess_em', 'scene', 'sun', 'radio_browser', 'stream', 'system_health', 'upnp', 'co2signal', 'auth', 'ocpp', 'persistent_notification', 'tplink', 'openweathermap', 'schedule', 'my', 'device_automation', 'wiz', 'image_upload', 'metoffice', 'map', 'application_credentials', 'template', 'file_upload', 'input_text', 'trace', 'ffmpeg', 'sensor', 'rest_command', 'logbook', 'zone', 'repairs', 'google_translate', 'octopus_energy', 'history', 'person', 'config', 'met'}
2023-12-20 10:54:55.141 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.foxess
2023-12-20 10:54:55.588 INFO (MainThread) [homeassistant.setup] Setting up foxess_em
2023-12-20 10:54:55.588 INFO (MainThread) [homeassistant.setup] Setup of domain foxess_em took 0.0 seconds
2023-12-20 10:54:55.844 INFO (MainThread) [custom_components.foxess_em] 
foxess_em
https://github.com/nathanmarlor/foxess_em/issues
2023-12-20 10:54:55.844 DEBUG (MainThread) [custom_components.foxess_em] Initialising Fox Cloud service
2023-12-20 10:54:56.169 ERROR (MainThread) [custom_components.foxess.sensor] Unexpected error fetching FoxESS data: Expecting value: line 1 column 1 (char 0)
  File "/config/custom_components/foxess/sensor.py", line 125, in async_update_data
  File "/config/custom_components/foxess/sensor.py", line 251, in authAndgetToken
2023-12-20 10:54:56.172 ERROR (MainThread) [custom_components.foxess.sensor] FoxESS Cloud initializaction failed, fix error and restar ha
2023-12-20 10:54:56.412 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.foxess_em
2023-12-20 10:54:56.424 INFO (MainThread) [homeassistant.components.number] Setting up number.foxess_em
2023-12-20 10:54:56.424 INFO (MainThread) [homeassistant.components.calendar] Setting up calendar.foxess_em
2023-12-20 10:54:56.425 INFO (MainThread) [homeassistant.components.switch] Setting up switch.foxess_em
2023-12-20 10:54:56.428 DEBUG (MainThread) [custom_components.foxess_em.fox] Logging into Fox Cloud
2023-12-20 10:54:56.428 DEBUG (MainThread) [custom_components.foxess_em.fox] Issuing request to (https://www.foxesscloud.com/c/v0/user/login) with params: {'user': 'Fingal', 'password': 'fe378b13422cd53a87ebcc396a1296ce'}
2023-12-20 10:56:28.349 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{((states('sensor.foxess_bat_soc') | float) - 10) / 100 * 8.06 }}) renders=2>
ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{((states('sensor.foxess_bat_soc') | float) - 10) / 100 * 8.06 }}' but no default was specified
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{((states('sensor.foxess_bat_soc') | float) - 10) / 100 * 8.06 }}' but no default was specified
2023-12-20 10:56:28.352 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{{((states('sensor.foxess_bat_soc') | float) - 10) / 100 * 8.06 }}' but no default was specified') while processing template 'Template<template=({{((states('sensor.foxess_bat_soc') | float) - 10) / 100 * 8.06 }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.battery_energy_remaining'
2023-12-20 10:56:28.364 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Forecast cache is 4:56:22.914721 old
2023-12-20 10:56:28.365 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Loading forecast data from cache
2023-12-20 10:56:28.458 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Finished loading forecast data from cache, notifying listeners
2023-12-20 10:56:28.458 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Refreshing Solcast site info
2023-12-20 10:56:28.458 DEBUG (MainThread) [custom_components.foxess_em.forecast] Retrieving API call status
2023-12-20 10:56:28.458 DEBUG (MainThread) [custom_components.foxess_em.average.average_controller] Refreshing averages model
2023-12-20 10:56:28.458 DEBUG (MainThread) [custom_components.foxess_em.battery.battery_controller] Refreshing battery model
2023-12-20 10:56:28.469 INFO (MainThread) [homeassistant.components.automation.fox_test_2] Initialized trigger Fox charge before 4
2023-12-20 10:56:28.469 INFO (MainThread) [homeassistant.components.automation.fox_new_new] Initialized trigger Fox new new
2023-12-20 10:56:28.469 INFO (MainThread) [homeassistant.components.automation.prevent_fox_going_below_10] Initialized trigger Prevent Fox going below 10
2023-12-20 10:56:28.469 INFO (MainThread) [homeassistant.components.automation.fox_keep_warm] Initialized trigger Fox keep warm
2023-12-20 10:56:28.469 INFO (MainThread) [homeassistant.components.automation.fox_too_hot] Initialized trigger Fox too hot
2023-12-20 10:56:28.572 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.foxess_em_forecast exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2023-12-20 10:56:28.598 WARNING (Recorder) [homeassistant.components.recorder.db_schema] State attributes for sensor.foxess_em_forecast exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
2023-12-20 10:56:28.714 DEBUG (MainThread) [custom_components.foxess_em.average.average_controller] Finished refreshing averages model, notifying listeners
2023-12-20 10:56:28.714 DEBUG (MainThread) [custom_components.foxess_em.battery.battery_controller] Refreshing battery model
2023-12-20 10:56:28.730 WARNING (MainThread) [custom_components.foxess_em.battery.battery_controller] Battery state is invalid
2023-12-20 10:56:29.290 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Clearing 0 refresh listeners
2023-12-20 10:56:29.291 DEBUG (MainThread) [custom_components.foxess_em.forecast] Retrieving available sites
2023-12-20 10:56:29.839 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Creating refresh schedule for 1 sites
2023-12-20 10:56:29.839 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Calculated 1 available refreshes
2023-12-20 10:56:29.839 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Setting up forecast refresh at 06:00:00
2023-12-20 10:56:29.840 DEBUG (MainThread) [custom_components.foxess_em.forecast.forecast_controller] Setting up forecast refresh at 17:59:29.291002
[... cut ...]
2023-12-20 11:02:00.323 DEBUG (MainThread) [custom_components.foxess_em.fox.fox_cloud_service] Requesting stop force charge from Fox Cloud
2023-12-20 11:02:00.323 DEBUG (MainThread) [custom_components.foxess_em.fox] Logging into Fox Cloud
2023-12-20 11:02:00.323 DEBUG (MainThread) [custom_components.foxess_em.fox] Issuing request to (https://www.foxesscloud.com/c/v0/user/login) with params: {'user': 'Fingal', 'password': 'fe378b13422cd53a87ebcc396a1296ce'}
2023-12-20 11:02:01.396 ERROR (MainThread) [custom_components.foxess_em.fox.fox_cloud_service] Could not make request to Fox Cloud - HTTP Status: 406

```

nathanmarlor, 2023-12-20:
@FingalP unfortunately Fox have introduced request/response signing in their latest cloud update which effectively excludes any third party integrations. There's a thread on this over on Facebook.

FingalP, 2023-12-20:
Damn, thanks for the quick reply! I discovered and requested to join the facebook group earlier today but I'm not in yet. My inverter is currently still force charging, is there a way to turn that off without using the integration? I can't find it in their UI.

nathanmarlor, 2023-12-20:
You can do it through the menu on the front of the inverter - looks for charge period settings.

closed 2024-02-21 by FingalP