nathanmarlor/foxess_em #350: Foxess EM not loaded

djdarlo, 2024-02-23:
## Version of the custom_component

## Configuration

```yaml
Add your logs here.
```

## Describe the bug

When looking at the integration in HA it is showing Not Loaded. I've attached a screenshot. [image]

## Debug log

```text

Add your logs here.

```

nathanmarlor, 2024-02-23:
Version? Logs? Working before? I have literally nothing to go off here...

djdarlo, 2024-02-23:
Apologies. Doing it from my phone as away. 

Home Assistant 
Core 2024.2.2
Supervisor 2024.02.0

Foxess EM 
V 1.8.0

It has worked previously. To make it work again. I deleted the integration. Reinstalled and configured again. It worked for two days then gave me the not loaded error again so I repeated the steps.