nathanmarlor/foxess_em #353: Updated foxess_em integration reports error for Bat SOC

ng34828, 2024-02-29:
First thanks to Nathan and Dan for updating the foxess_em integration to use the public API

Configured it this morning and I am seeing the following error in the log when I used the Bat - SOC entity from the foxess integration:

2024-02-29 09:06:34.068 ERROR (MainThread) [custom_components.foxess_em.battery.battery_controller] ValueError("invalid literal for int() with base 10: '17.0'")
2024-02-29 09:06:35.986 ERROR (MainThread) [custom_components.foxess_em.battery.battery_controller] ValueError("invalid literal for int() with base 10: '17.0'")

and when I looked the entities provided by foxess_em most were in the Unknown state

Work round was to use a template to create a new Bat SOC entity which was an integer - having configured that into the integration it all seems to be working OK

siobhanellis, 2024-05-10:
Still have this problem

siobhanellis, 2024-05-28:
If it makes any difference, I'm using the Foxess Cloud integration.