nathanmarlor/foxess_em #354: Migration failed error

screenagerbe, 2024-03-01:
## Version of the custom_component

1.9.0

## Configuration

```yaml
Modbus via TCP
```

## Describe the bug
After upgrading from version 1.8.0 and restarting HA the integration would not load. It said _Migration Failed_ on the integrations page and all values had status `unknown`.
Solcast was also down.

In the HA logs I saw this "Warning"
[image]

## Workaround 

```text
Remove the integration
Restart HA
Reinstall the integration
Restart HA
Add and configure the integration
```

djdarlo, 2024-03-03:
Got the same - will try the workaround and let you know how I get on.

danelphick, 2024-03-04:
Sorry about this. This was my fault. It's fixed in https://github.com/nathanmarlor/foxess_em/pull/356.

You can also fix this without removing and re-added, by sshing into your HA and editing config/.storage/core.config_entries

Search for the foxess_em section and change:

"version": 1
to
"version": 2

You'll need to restart HA for this to take effect. You still need to reconfigure the integration either via the UI or you can add a line in the file above

"fox_api_key": "<insert your key here>"

making sure to observe JSON rules about trailing commas.

nathanmarlor, 2024-03-04:
Try v1.9.1 please :)

djdarlo, 2024-03-04:
Will do ;)