nathanmarlor/foxess_em #367: Forecast no longer updating automatically

andrewmooreio, 2024-05-27:
## Version of the custom_component

v1.9.1
HA Core 2024.05.1

## Configuration

N/A

## Describe the bug

Since the 18th May, the forecast has stopped updating automatically.

I have to reload the integration or restart HA to generate a new forecast.

I've reinstalled the integration, which didn't fix the issue.

## Debug log

When I restart the integration I get this error but nothing else.
```text

TypeError("'NoneType' object is not subscriptable")

```

andrewmooreio, 2024-05-29:
Strangely, the forecast graph does update but the "forecast_today" and "forecast_tomorrow" entities don't.

browser801, 2024-05-30:
Having the exact same issue

screenagerbe, 2024-06-01:
same error, but the graph doesn't update either.
got three errors in my logs coming from this integration:

Errors when the daily refresh gets triggered (this one comes up 2 times at the same time, I reckon for each PV array configured at SolCast)

```
Logger: homeassistant
Source: custom_components/foxess_em/battery/battery_controller.py:128
integration: FoxESS - Energy Management ([documentation](https://github.com/nathanmarlor/foxess_em), [issues](https://github.com/nathanmarlor/foxess_em/issues))
First occurred: 31 May 2024 at 00:14:01 (251 occurrences)
Last logged: 16:42:02
Error doing job: Task exception was never retrieved

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 968, in async_update_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1132, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1015, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 543, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/foxess_em/common/sensor.py", line 66, in native_value
    return method()
           ^^^^^^^^
  File "/config/custom_components/foxess_em/battery/battery_controller.py", line 128, in state_at_eco_start
    return round(self._schedule_info()["battery"], 2)
                 ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

Error after reloading the integration:

```
Logger: custom_components.foxess_em.forecast.forecast_controller
Source: custom_components/foxess_em/forecast/forecast_controller.py:180
integration: FoxESS - Energy Management ([documentation](https://github.com/nathanmarlor/foxess_em), [issues](https://github.com/nathanmarlor/foxess_em/issues))
First occurred: 17:12:08 (1 occurrences)
Last logged: 17:12:08
TypeError("'NoneType' object is not subscriptable")
```