nathanmarlor/foxess_em #383: Overestimated solar frecast since update

kme77, 2024-11-02:
Core 2024.10.4
Foxess_em V.1.9.1

I hadn't updated my setup for a few months as everything was working fine. Decided to update it last week 29/10 and since then the solar forecast is WILDLY overestimating the forecasted solar production and as a result my battery has been depleted and I've had to manually charge overnight.

I've attached a screenshot of the graph showing solar forecast and actual solar production for the last week or so. As you can see from the 30th it's WAY off.

Any help would be appreciated.

I'm terrible with HA so I don't have logs but if required I can try to get them.

Thanks.
[image]

icornish72, 2024-11-02:
I have a very similar experience in recent days - the forecast data is not so great and the very changeable local weather makes it worse.
Not a lot you can do, except keep an eye on things

kme77, 2024-11-02:
> I have a very similar experience in recent days - the forecast data is not so great and the very changeable local weather makes it worse. Not a lot you can do, except keep an eye on things

Thanks for the reply, I HOPE it's just a temporary issue with the Solcast data maybe?
 I'm just very suspicious as the forecast data has always been pretty reasonable and produced good results for me over the last 2 years. It's only literally since I updated that the predicted solar data is WELL off (over double what's actually produced) which makes me lean towards an issue with how the update affected the integration.

andrewmooreio, 2024-11-02:
This is not an issue with the integration, just an inaccurate forecast from Solcast. I use the "Solcast PV Forecast" integration and the results are the same. You can also poll the Solcast API directly to get the raw data if you're unsure.

Predicting solar output is incredibly difficult, you have to model cloud cover to insane levels of accuracy. Sometimes certain weather patterns just make it next-to-impossible to get right. Solcast are arguably the best in the world at it, which just demonstrates how hard it is.

kme77, 2024-11-02:
> This is not an issue with the integration, just an inaccurate forecast from Solcast. I use the "Solcast PV Forecast" integration and the results are the same. You can also poll the Solcast API [...]

Thanks for the reply. As I said I hope that this IS the issue but you can understand why I want to check as I've not encountered such a huge difference in expected vs realised output in the entire time I've used it and the issue started literally the day after I updated everything.

I normally check the Solcast site to verify on their chart, but could you tell me how to poll the API directly?

Cheers ☺️