nathanmarlor/foxess_em #387: Conflict with FoxESS-Modbus

screenagerbe, 2025-01-06:
After updating FoxESS-Modbus to v1.13.0 and HACS to v2.0.2 we run into issues because of different versions in pymodbus.

See also https://github.com/nathanmarlor/foxess_modbus/issues/722
More specific: https://github.com/nathanmarlor/foxess_modbus/issues/722#issuecomment-2571430130

pmviana, 2025-01-09:
Yes, I can confirm. Uninstalled FoxESS-Modbus v1.13.0 and installed previous version. It's possible to bump up pymodbus?

canton7, 2025-02-04:
foxess_modbus now uses its own isolation version of pymodbus so won't conflict, but you will hit problems with HA 2025.1 and above which uses an incompatible pymodbus version.

screenagerbe, 2025-02-04:
> foxess_modbus now uses its own isolated version of pymodbus so won't conflict, but you will hit problems with HA 2025.1 and above which uses an incompatible pymodbus version.

Aah that's too bad.
I was only using the solcast integration because it is more accurate than the default integration provided by HA.
Seems I'll have to do it with the default for now.

Turak64, 2025-08-16:
> Aah that's too bad. I was only using the solcast integration because it is more accurate than the default integration provided by HA. Seems I'll have to do it with the default for now.

I've done it and restarting HA doesn't show the error anymore. I can see the force charge actions still, so will test it later to confirm it works.