nathanmarlor/foxess_em #397: Not Loaded on Core 2025.8.2

nuttersrest, 2025-08-16:
After updating to Core 2025.8.2 FoxESS-EM fails to load, looking at the error log it is missing a specific version of pymodbus.

Currently running 
foxess_em 1.10
 
On 

Installation method: Home Assistant OS
Core:            2025.8.2
Supervisor:  2025.08.1
Operating System:    16.1
Frontend:      20250811.0

FoxESS-Modbus 1.14.0b3 is running fine on the same machine

Logger: homeassistant.setup
Source: setup.py:278
First occurred: 01:23:28 (1 occurrence)
Last logged: 01:23:28

Setup failed for custom integration 'foxess_em': Requirements for foxess_em not found: ['pymodbus==3.1.3']

Turak64, 2025-08-16:
I've now updated the manifest.json file to the below, after restarting HA, this seemed to have fixed the issue. ChatGPT has also checked the pymodbus version upgrade and it should be compatible 

`{
  "domain": "foxess_em",
  "name": "FoxESS - Energy Management",
  "codeowners": ["@nathanmarlor"],
  "config_flow": true,
  "dependencies": ["recorder"],
  "documentation": "https://github.com/nathanmarlor/foxess_em",
  "integration_type": "service",
  "iot_class": "cloud_polling",
  "issue_tracker": "https://github.com/nathanmarlor/foxess_em/issues",
  "requirements": ["pymodbus>=3.1.3,<4.0.0", "pandas"],
  "version": "0.0.3"
}`

Turak64, 2025-08-16:
Great news—based on the release notes and changelogs, upgrading to PyModbus 3.9.2 should be smooth, with no breaking API changes from version 3.1.3. The project follows a custom versioning policy:

Patch-level updates (e.g., 3.9.0 → 3.9.2) do not introduce API changes, only bug fixes or enhancements.

The previous concern about dependency conflicts is resolved by relaxing the requirement in your manifest.json.

---

What's Changed in PyModbus (3.1.3 → 3.9.2)

Change Type	Details

3.9.2 (Patch)	No API changes—safe to upgrade
3.9.0 (Minor)	Bit handling (e.g. read_coils) was corrected; some method names updated
3.10.0 (Minor)	ModbusSlaveContext replaced with ModbusDeviceContext; parameters slave= became device_id=; bit return order adjusted

Conclusion: If your integration doesn’t depend on legacy bit-order behavior or outdated context class names, it should work fine with 3.9.2.

---

Updated manifest.json

Ensure your custom_components/foxess_em/manifest.json looks like this:

{
  "domain": "foxess_em",
  "name": "FoxESS - Energy Management",
  "codeowners": ["@nathanmarlor"],
  "config_flow": true,
  "dependencies": ["recorder"],
  "documentation": "https://github.com/nathanmarlor/foxess_em",
  "integration_type": "service",
  "iot_class": "cloud_polling",
  "issue_tracker": "https://github.com/nathanmarlor/foxess_em/issues",
  "requirements": ["pymodbus>=3.1.3,<4.0.0", "pandas"],
  "version": "0.0.3"
}

This allows Home Assistant to install the already-present 3.9.2, satisfying the requirement.

---

Post-Update Checklist

1. After updating the manifest, restart HA. Ensure foxess_em loads without dependency errors.

2. Monitor logs for runtime errors tied to pymodbus.

3. If needed, patch your code:

Replace ModbusSlaveContext with ModbusDeviceContext.

Rename slave= to device_id=.

Adjust bit-reading logic if coils are interpreted differently.

---

Would you like help checking your existing Modbus client code for these updates—or even preemptively patch it to be resilient for 3.10+? I’d be happy to assist.

nuttersrest, 2025-08-16:
Sadly this wont work, it allows the error notification to disappear but foxess-em will then fail to load.
The reason it fails to load is that the code is not compatible with newer versions of pymodbus, see the following [#387](https://github.com/nathanmarlor/foxess_em/issues/387). So to my limited knowledge of this means that unless someone is able to update the foxess-em code for support on newer versions of pymodbus then it will not be able to work on systems with Core 2025.8.2 or greater.

andy-hawkes-deltatre, 2025-08-21:
I've run some changes via Copilot to upgrade to pymodbus 3.9.2 and resolve breaking changes - looks quite minor so I'm testing locally.
It seems OK so far but I want to leave it overnight to see how it manages the cheap charge period (where it should not need to charge and should set the minSOC accordingly) - if that's all good I'll share the changes tomorrow for anyone that is interested.

andy-hawkes-deltatre, 2025-08-22:
I'm not really a Python dev, so it's a bit hit-and-miss, but it looks like my changes were OK - MinSOC On Grid was managed properly overnight.

Pymodbus >= 3.8 enforces keyword-only parameters, so this needed to be handled. Do not upgrade beyond 3.9 without further changes (3.10 introduced device_id=).

I'm in the middle of a bunch of work stuff so I don't have time to create a pull request right now, but I've created a zip file of the changed files if anyone wants to take a look - [pymodbus-392-fixes.zip](https://github.com/user-attachments/files/21934389/pymodbus-392-fixes.zip)

Turak64, 2025-08-22:
Quick one, it doesn't appear to be accepting my Api key, even though I know it's correct. Had any luck with that?

Turak64, 2025-08-24:
I only use this intergration for the force charging start/stop actions and those are now working. However, I did see this issue today.
```

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:350
First occurred: 04:00:01 (1 occurrence)
Last logged: 04:00:01

Detected that custom integration 'foxess_em' calls `async_track_state_change` instead of `async_track_state_change_event` which is deprecated and will be removed in Home Assistant 2025.5 at custom_components/foxess_em/charge/charge_service.py, line 192: track_charge = async_track_state_change(. Please create a bug report at https://github.com/nathanmarlor/foxess_em/issues

This error originated from a custom integration.
```

Why this is deprecated

async_track_state_change is being phased out.

HA recommends using async_track_state_change_event, which provides the same functionality but in the new event-based format.

The signature is slightly different:

`async_track_state_change_event(hass, entity_ids, async_callback)
`
It passes an event object to your callback instead of entity, old_state, new_state.

How it should be updated

Replace the listener setup with:

from homeassistant.helpers.event import async_track_state_change_event
```

def _start_listening(self):
    """Listen for SOC changes using the new event-based API."""
    track_charge = async_track_state_change_event(
        self._hass,
        self._battery_soc,
        self._battery_soc_change_event
    )
    self._cancel_listener = track_charge
    self._unload_listeners.append(track_charge)

```
Adjust the callback to accept an event:
```
async def _battery_soc_change_event(self, event):
    """Handle SOC changes from async_track_state_change_event."""
    try:
        new_state = float(event.data.get("new_state").state)
    except Exception:
        return

    if self._custom_charge_profile and new_state > 90:
        step_down_charge = round(((100 - new_state) / 10) * self._user_charge_amps, 2)
        target_charge_amps = max([_MINIMUM_CHARGE, min([step_down_charge, self._target_charge_amps])])
        await self._fox.set_charge_current(target_charge_amps)

    if (new_state >= self._perc_target) and self._charge_active:
        if self._perc_target != 100:
            await self._stop_force_charge()
    elif (new_state < (self._perc_target - _CHARGE_HYSTERESIS) and not self._charge_active):
        await self._start_force_charge_off_peak()

```
✅ Summary

Replace async_track_state_change with async_track_state_change_event.

Update the callback to accept an event object and extract new_state from event.data.

This resolves the deprecation warning and ensures compatibility with HA 2025.5+.

Turak64, 2025-08-24:
------------------

Then I also had this issue

```

Logger: homeassistant
Source: custom_components/foxess_em/battery/battery_controller.py:141
integration: FoxESS - Energy Management (documentation, issues)
First occurred: 03:55:04 (1 occurrence)
Last logged: 03:55:04

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/foxess_em/charge/charge_service.py", line 106, in _eco_start_setup
    self._charge_required = self._battery_controller.charge_total()
                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/foxess_em/battery/battery_controller.py", line 141, in charge_total
    return self._schedule_info()["total"]
           ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

The main error you were seeing in the logs is:
```

TypeError: 'NoneType' object is not subscriptable
File ".../battery_controller.py", line 141, in charge_total
return self._schedule_info()["total"]
```

Looking at the _schedule_info() method:
```
def _schedule_info(self) -> float:
    """Schedule info"""
    return self._schedule.get(self._peak_utils.next_eco_start())
```

_schedule.get(...) can return None if there is no schedule for the computed next_eco_start().

Your other methods (charge_total, dawn_charge_needs, etc.) then assume the return is always a dictionary and subscriptable.

This causes TypeError: 'NoneType' object is not subscriptable when _schedule.get() returns None.

Recommended Fix

Wrap _schedule.get() with a fallback to ensure it always returns a dictionary, even if the schedule is missing:
```

def _schedule_info(self) -> dict:
    """Schedule info"""
    info = self._schedule.get(self._peak_utils.next_eco_start())
    if info is None:
        _LOGGER.warning("Schedule missing for next eco start, returning default empty schedule")
        info = {"battery": 0, "dawn": 0, "day": 0, "total": 0, "min_soc": 0}
    return info

```

This prevents NoneType errors.

Maintains all existing calls like charge_total(), dawn_charge_needs(), etc.

Logs a warning so you can investigate why the schedule is missing.

andy-hawkes-deltatre, 2025-09-02:
Looks like my fix may have broken again on 2025.8.3 - the battery has charged fully for the last 2 nights since I upgraded. Time to dig into the logs again...

jonmunson, 2025-12-30:
I've created a fix, tested working on 2025.12.5 - pull request here: https://github.com/nathanmarlor/foxess_em/pull/399